leka/LekaOS #301: [Task] - Vibrations demo

SamHadjes, 2021-07-21:
## Create a base to test and implement the haptic functionality of Leka

## Subtasks 
## Subtasks 

## Subtasks 

- [ ] 
- [ ]
- [ ]

Lucieveber, 2021-07-21:
Pour **Count vibrations**, this is repeated in a **random** order yes !

Lucieveber, 2021-07-21:
Pour **Different feels with Leka**, vous voyez ça comment ? Des séquences prédéfinies déjà (genre 4-5) ou bien carrément une commande permettant au pro/à l'enfant de créer leur propre séquence en mode manuel en choisissant chacune des variables. La manière dont est tourné le paragraphe "Different feels with Leka" ça donne cette impression "mode manuel" (vs automatique). C'est pas ce que j'imaginais au début mais si c'est possible de faire ça c'est top !

YannLocatelli, 2021-07-21:
About the envelop, we already talked about it: it is intrinsically linked to duration/length and amplitude.

You can focus first on frequency and amplitude changed on the fly and length. 
Envelop can be considered as variations of amplitude in function of time and record it in a file, but we mentioned another approach: generate it by computing it from a formulae. Until you have done the comparison of both methods, this parameter is not a priority.

Aermanio, 2021-07-21:
> Pour **Different feels with Leka**, vous voyez ça comment ? Des séquences prédéfinies déjà (genre 4-5) ou bien carrément une commande permettant au pro/à l'enfant de créer leur propre séquence en [...]

Aermanio, 2021-07-21:
> Pour **Different feels with Leka**, vous voyez ça comment ? Des séquences prédéfinies déjà (genre 4-5) ou bien carrément une commande permettant au pro/à l'enfant de créer leur propre séquence en [...]

On peut envisager dans un premier temps de faire des séquences prédéfinies, mais à notre niveau ce n'est pas si compliqué de rendre possible la création d'une vibration dont la fréquence/durée/intensité est entrée en commande, surtout si on ne se complique pas la tâche avec des enveloppes/formes particulières pour le moment. 
Après, on ne créerait que la fonction permettant de lancer cette vibration personalisée, pour la partie interfaçage utilisateur et intégration avec l'application ou autre ce n'est pas nous qui nous en occuperons.

ladislas, 2022-01-24:
see #453