leka/LekaOS #385: [Task] - Activities for launch day

ladislas, 2021-11-21:
## Description

Several activities to do on launch day

**Related epic/story/task**:  n/a

## v1.0

### Reconnaitre la couleur

- Leka affiche une image ayant une couleur dominante (maison, poisson, sceau, ballon)
- la personne doit apporter le tag RFID avec la couleur correspondante
- en cas de bonne réponse, Leka se colore dans la couleur reconnue puis lance un renforçateur (soit le rainbow, soit aléatoire)
- Leka s'éteint, l'écran aussi pendant 1 seconde
- puis une nouvelle image s'affiche
- les images peuvent faire partie de la même famille (que les maisons) ou être mélangées (poisson puis maison, puis sceau, etc.)
- en cas d'erreur il ne se passe rien

### Reconnaitre les émotions

- Leka affiche une émotion sur son écran
- la personne doit apporter le tag RFID présentant la même émotion
- le tag peut être un tag avec Leka ou un tag avec un visage d'enfant
- en cas de bonne réponse, Leka lance un renforçateur (soit le rainbow, soit aléatoire)
- Leka s'éteint, l'écran aussi pendant 1 seconde
- puis une nouvelle image s'affiche
- en cas d'erreur il ne se passe rien

### Reconnaitre les chiffres

- même principe qu'avant, mais avec les chiffres de 1 à 10.

### Reconnaitre les formes

- même principe qu'avant, mais avec les formes pleines ou pointillées

### Faire apparaitre une émotion

- Leka affiche un visage neutre
- quand la personne approche un tag RFID avec une émotion, Leka affiche la même émotion pendant 4 secondes
- puis Leka redevient neutre
- si une autre émotion est posée pendant que la première est affichée, rien ne se passe et elle est oubliée

### Colorer Leka 1

- Leka affiche un visage neutre
- quand la personne approche un tag RFID présentant une couleur, Leka se colore de la même couleur et affiche le visage heureux pendant 4 secondes
- puis Leka redevient neutre
- si une autre couleur est posée pendant que la première est affichée, rien ne se passe et elle est oubliée

### Colorer Leka 1

- Leka affiche un visage neutre
- quand la personne approche un tag RFID présentant une couleur, Leka se colore de la même couleur et change d'expression, devenant heureux
- Leka garde l'expression et la couleur sans limite
- un autre tag doit être approché pour changer de couleur
- l'émotion change aussi et alterne entre heureux et neutre

### Lancer un renforçateur

- Leka affiche un visage neutre
- quand la personne approche un tag RFID présentant un renforçateur, Leka joue le renforçateur et affiche un visage heureux
- à la fin de l'animation, Leka redevient neutre

## v2.0

### Calculette

- Leka affiche un nombre entre 1 et 10
- la personne doit apporter 2 nombres dont l'addition est égale au nombre affiché
- en cas d'erreur, le nombre affiché change
- en cas de réussite, Leka est heureux et un renforçateur est joué

### Couleur + forme

- Leka affiche une forme colorée
- la personne doit apporter la carte de la forme et la carte de la couleur représentées
- en cas d'erreur, la forme affichée change
- en cas de réussite, Leka est heureux et un renforçateur est joué

YannLocatelli, 2021-11-22:
## Suivi réalisations

## v1.0

-[ ] Reconnaitre la couleur
-[ ] Reconnaitre les émotions
-[ ] Reconnaitre les chiffres
-[ ] Reconnaitre les formes
-[ ] Faire apparaitre une émotion
-[ ] Colorer Leka 1
-[ ] Colorer Leka 1
-[ ] Lancer un renforçateur

## v2.0

-[ ] Calculette
-[ ] Couleur + forme

ladislas, 2021-11-28:
@YannLocatelli j'ai mis à jour pour les transitions, tu me diras si ça te va.

ladislas, 2021-12-13:
Well done guys! 🎉

closed 2021-12-13 by ladislas