leka/LekaOS #453: Previous Work - Audio, DAC, Vibrations

ladislas, 2022-01-24:


YannLocatelli, 2022-01-25:
## Comments

### DACDriver and DACTimer #302

AnalogOut of mbed cannot use the DMA, a tool that can transfer data from a port to another (here from SRAM to DAC output) and allow to free CPU from this task. DACDriver and DACTimer implement an "AnalogOut" using the DMA with ST DAC driver only.

DAC output is used to vary tension at a specific rate to generate waves of vibrations & sounds into haptic vibrator. To synchronise the output at a specific rate, a Timer is used and due to the nature of DAC (implement ST driver and not mbed), the Timer is also from ST and its configuration vary from mbed Timer. There are only 2 available for DAC & DMA and their max frequency can go from [1.6KHz to 108MHz (max of the MCU)]. DACTimer wrap the ST Timer to fit with ST DAC to allow it to output data at a specific rate.

DACDriver is a wrapper of ST DAC. It uses DACTimer and also a DMA. Apart from hid configuration, DMA can notify callback when the last data of first and second half of its internal buffer are transferred . This principle is used in spike to call `callbackTest` and also later on to load next data of sound or vibration.

DACTimer and DACDriver are fully implemented, not unit tested and have to be renamed to fit with Leka’s style. The spike associated is `lk_dac`.

**Complementary documents**: [[Reference Manual](https://www.st.com/resource/en/reference_manual/dm00224583-stm32f76xxx-and-stm32f77xxx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf#page=486)]

### CoreAudio

WAVFile is a FIL wrapper and consider file as WAV format and gather informations in header of the file (RIFF).

WAVReader convert data stored in file (WAVFile) into correct format → Data is stored in `uint8_t`, data sound is encoded in `int16_t` in mono. WAVReader do the transfer from SD to SRAM.

CoreAudio uses WAVReader to get sound data and DACDriver to play sound data. It also controls the volume. It also refill the DMA buffer of DACDriver when it reach the first or second half.

There is a high-pass filter at 100 Hz to avoid the resonance at 60 Hz. Most of suggestion can be find in related issue #303. One remaining task is to set DACTimer and DACDriver internal to CoreAudio.

CoreAudio is fully implemented but not unit tested. The spike associated is `lk_audio`.

### CoreVibration #306

An envelope is applied on a constant frequency to modulate amplification and render a specific vibration.

The resonance [[Link](https://en.wikipedia.org/wiki/Resonance)] is not handled.

CoreVibration is considered as an advanced draft.

ladislas, 2022-01-25:
Thanks @YannLocatelli for the details! :)