leka/LekaOS #643: BLE Connection - find a way to know which Leka is which --> display a color

ladislas, 2022-04-04:
on first boot, leka is blue
when the user configures Leka for the first time, they are asked to choose a custom name for their Leka and a custom color.
this allows the user to always know which Leka they are using and to easily choose amongst multiple Leka

YannLocatelli, 2022-05-20:
**Update.**

Since the BLE module is Bluetooth v4.2 [[Link](https://www.st.com/en/wireless-connectivity/bluenrg-m0.html)], some features are not available, including notification on scan.

The feature dealing with `onScanRequestReceived` [[Link](https://github.com/ARMmbed/mbed-os/blob/1ab98dec9ab4d07433d6eaec84cc4049a4a7b56b/connectivity/FEATURE_BLE/include/ble/Gap.h#L304-L315)] is `LE_EXTENDED_ADVERTISING`. It is possible to check if it is available with the following call
```cpp
_gap.isFeatureSupported(ble::controller_supported_features_t::LE_EXTENDED_ADVERTISING); // In CoreGap.cpp
```

Note. Some other actions seems to be necessary as explained here: [[AdvertisingParameters](https://github.com/ARMmbed/mbed-os/blob/1ab98dec9ab4d07433d6eaec84cc4049a4a7b56b/connectivity/FEATURE_BLE/include/ble/gap/AdvertisingParameters.h#L71-L72)] and of course use of [`setScanResponse`](https://os.mbed.com/docs/mbed-os/v6.15/mbed-os-api-doxy/classble_1_1_gap.html#a7233000dd08110c333cc076291a72359)

YannLocatelli, 2022-05-23:
Maybe use another way to interact with robot when not connected: IMU with a tap, Touch sensor, Microphone, a mix of all of this, etc.

ladislas, 2022-05-30:
Moved to `Future releases` milestone https://github.com/leka/LekaOS/milestone/5

This will give us time to find a different solution.