leka/LekaOS #656: RC - Constructor use a struct

YannLocatelli, 2022-04-05:


ladislas, 2022-04-18:
I've thought about that this weekend and I'm wondering if it will really improve readability/usability.

The issue with using a struct is that everywhere in RC's code we'll have things like `_struct_name.battery.isCharging()` or `_struct_name.videokit.playAnimation()` etc.

It will make RC's code harder to read.

On `os/main.cpp` side, it will not make much of a difference. it will just introduce a new variable to pass to RC's constructor, but the variable will need to be created the same way.

One way to fix that would be to make a factory function creating RC, but that usually mean heap, and that's not good.

boost::di could be a good solution, we could investigate.

In any case, this is not urgent, so I'm removing it from v1.0.0

ladislas, 2022-06-22:
closing for now as not important

closed 2022-06-22 by ladislas