leka/LekaOS #863: Discussion - Watchdog in bootloader

HPezz, 2022-06-10:


YannLocatelli, 2022-06-10:
I can't see the direct benefit of a watchdog to reboot the bootloader

Same for the logger for a released product and not for a serial linked debugged product

Let me know what are your motivations

ladislas, 2022-06-15:
Motivations from slack: https://weareleka.slack.com/archives/C04F19WD6/p1655127187185149

> # bootloader + watchdog
> 
> si par un hasard et/ou un rassemblement de conditions et/ou un bug dans mcuboot, on se retrouve coincer dans le bootloader, il n’y a aucun moyen d’en sortir à moins d’attendre 24h que le robot se décharge.
> 
> même si la condition est inconnue, et que son éventualité est très faible, la risque n’est pas nul et l’impact est assez négatif pour l’utilisateur et pour nous.
> 
> le watchdog permet de palier ce risque très facilement et d’être future proof.
> 
> j’ai peut être loupé quelque chose, mais a priori je ne vois pas de drawbacks particuliers à l’implémenter.
> 
> mais si tu en as, vu que tu es beaucoup plus expert que nous sur le sujet du bootloader, je suis tout ouïe :slightly_smiling_face: