leka/LekaOS #864: Discussion - Battery level representation

HPezz, 2022-06-10:


YannLocatelli, 2022-06-10:
And what is displayed between 0% and 5%?

ladislas, 2022-06-15:
Adding here what I was saying there: https://github.com/leka/LekaOS/pull/870#discussion_r897353150

## Initial idea

### < 5 et < 25

[image]

### < 50

[image]

### < 75

[image]

### < 90

[image]

### > 90

[image]

ou 

[image]

ladislas, 2022-06-15:
## Alternative idea

### < 5 et < 25

[image]

### < 50

[image]

### < 75

[image]

### < 85

[image]

### < 95

[image]

### > 95

[image]

closed 2022-06-15 by ladislas