leka/LekaOS #1335: Update documentation

Alex0vSky, 2023-10-13:
Hello **LekaOS** team!

I'm collecting a list of projects that use state machines.
If we can see in the documentation that **LekaOS** uses the `boost-ext/sml` library then it will be easier for people to find and meet.
I wanted to help with this, quickly make a fork and fix the documentation and make PR...
But I don't see any list of dependencies or third parties.

Maybe I should start by making a list of dependencies?

ladislas, 2023-10-16:
Hi @Alex0vSky --

Thanks for the idea and for willing to contribute.

> Maybe I should start by making a list of dependencies?

Yes, that would be great. Then we can update the README with a section at the end.

Some deps from the top of my head:

- [mbed-os](https://github.com/armmbed/mbed-os/)
- [boost-ext/sml](https://github.com/boost-ext/sml)
- https://github.com/google/googletest
- https://os.mbed.com/users/JohnnyK/code/mbed-http/
- https://github.com/mcu-tools/mcuboot

Let me know if you have any questions

Alex0vSky, 2023-10-19:
Greetings @ladislas!

This is what we have as a result of the search.

A simple search of keywords and "include" yielded little:
[boost-ext/ut](https://github.com/boost-ext/ut)

If we look at cmake files: can not see anything.
If we look at LekaOS/tests: dont see anything new.
If we look at LekaOS/libs: only internal development.

Much more results when searching for the `LekaOS/*external` directory:

1. LekaOS/libs/IMUKit/external = could not find the source code for `Fusion library` from `Sebastian Madgwick`
2. LekaOS/drivers/CoreVideo/external/include/external + LekaOS/drivers/CoreVideo/external = https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Utilities/JPEG/jpeg_utils.c
3. LekaOS/drivers/CoreQDAC/include/external = lost, from dropbox.com
4. LekaOS/drivers/CoreIOExpander/include/external = unknown, unclear
5. LekaOS/drivers/CoreHTS/external = is this also "mbed"?

And look at regexp `Copyright (?!\d\d\d\d(-2022)? APF)` in "*.cpp;*.h;*.c;": 
"STMicroelectronics" and "ARM Limited" only, excluding "mbed-http"

This is all I was able to find in addition to what you wrote above.

YannLocatelli, 2023-11-07:
Hi @Alex0vSky !

Thank you again for willing to contribute !

In addition of the list given by @ladislas, you have indead spotted a major external library: [boost-ext/ut](https://github.com/boost-ext/ut)

I will now answer your different points:

> 1\. LekaOS/libs/IMUKit/external = could not find the source code for Fusion library from Sebastian Madgwick

The algorthim is from Sebastian Madgwick but the **application** of the algorithm is from xioTechnologies  
→ [xioTechnologies/Fusion](https://github.com/xioTechnologies/Fusion)

> 2\. LekaOS/drivers/CoreVideo/external/include/external + LekaOS/drivers/CoreVideo/external = https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Utilities/JPEG/jpeg_utils.c

It is from STMicroelectronics, and you already find it  
→ [https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Utilities/JPEG/jpeg_utils.c](https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Utilities/JPEG/jpeg_utils.c)

> 3\. LekaOS/drivers/CoreQDAC/include/external = lost, from dropbox.com

This "external code" is not an existing code but written by us based on the datasheet of the component [[Link](https://ww1.microchip.com/downloads/en/devicedoc/22187e.pdf)]  
→ X

> 4\. LekaOS/drivers/CoreIOExpander/include/external = unknown, unclear

Same as above, written by us based on the datasheet of the component [[Link](https://ww1.microchip.com/downloads/en/devicedoc/20001952c.pdf)]  
→ X

> 5\. LekaOS/drivers/CoreHTS/external = is this also "mbed"?

Same as number 2. It is from STMicroelectronics  
Intially from [STMems\_Standard\_C\_drivers](https://github.com/STMicroelectronics/STMems_Standard_C_drivers) repo but now available in a specific one  
→ [STMicroelectronics/hts221-pid](https://github.com/STMicroelectronics/hts221-pid)

---

Finally, you miss some:

6\. LekaOS/drivers/CoreIMU/extern, same as 5.
Intially from [STMems\_Standard\_C\_drivers](https://github.com/STMicroelectronics/STMems_Standard_C_drivers) repo but now available in a specific one  
→ [STMicroelectronics/lsm6dsox-pid](https://github.com/STMicroelectronics/lsm6dsox-pid)

7\. LekaOS/include/semver/semver.hpp
→ [Neargye/semver](https://github.com/Neargye/semver)

Let us know if you have any further question!

Alex0vSky, 2024-01-14:
Everything is fine, everything is ready, thank you @YannLocatelli  and @ladislas.
I am ready to make *PullRequest*, I'm not sure where we should put the text...
Maybe here?
`LekaOS/extern/README.md`

Here is the full list, in order of discussion and appearance:
[armmbed/mbed-os](https://github.com/armmbed/mbed-os/)
[boost-ext/sml](https://github.com/boost-ext/sml)
[google/googletest](https://github.com/google/googletest)
https://os.mbed.com/users/JohnnyK/code/mbed-http/
[mcu-tools/mcuboot](https://github.com/mcu-tools/mcuboot)
[boost-ext/ut](https://github.com/boost-ext/ut)
[LekaOS/libs/IMUKit/external](https://github.com/xioTechnologies/Fusion)
[STMicroelectronics/STM32CubeF7](https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Utilities/JPEG/jpeg_utils.c)
[STMicroelectronics/hts221-pid](https://github.com/STMicroelectronics/hts221-pid)
[STMicroelectronics/lsm6dsox-pid](https://github.com/STMicroelectronics/lsm6dsox-pid)
[Neargye/semver](https://github.com/Neargye/semver)

Alex0vSky, 2024-01-15:
I chose the name of the new section using the most stellar resource [jehna/readme-best-practices](https://github.com/jehna/readme-best-practices).
The choice was between "Credits|Links|Acknowledgments|Thanks".
Still, I decided to leave what was in my first message: *"List of dependencies"*.
Please correct me if I missed something. Thank you.

Alex0vSky, 2024-01-16:
Closed by 4c5dfca via PR [1355](https://github.com/leka/LekaOS/pull/1355)
Many thanks!

closed 2024-01-16 by Alex0vSky